hlorus/CAD_Sketcher #405: [BUG] Extra lines on my mesh

ClayShoaf, 2023-07-15:
### Description

For some reason (maybe I have too many constraints, maybe I don't have enough) sometimes when I am exiting a sketch, there will be and extra line (plane? Sorry, I don't know all of the terminology) as you can see in the image on the left side of the "M". 
[image]

This is causing problems when I try to do a bool operation.

If I solidify it, it creates all kinds of crazy faces that you can see in wireframe view:
[image]

I have been able to make that division go away by playing around with deleting constraints, but there doesn't seem to be any logic in which constraints cause it to happen.

I'm just wondering if this is a bug, or if I'm doing something wrong.

### Addon Version

0.27.2

### Blender Version

3.6.0

### What platform are you running on?

Linux

hlorus, 2023-09-07:
This comes from the fact that blender is mesh-based. The addon converts the geometry to blenders bezier curves, the extra lines then come from blenders bezier-mesh converter. You should be able to remove them with a decimate modifier.

closed 2023-09-07 by hlorus

ClayShoaf, 2023-09-09:
But then why does it not happen if I remove the constraint and move the point a little bit?