hlorus/CAD_Sketcher #406: [BUG]

IknietBegrijp, 2023-07-21:
### Contact Details

/

### Description

I just installed CAD sketcher for blender, but now when i try to add a sketch no construction axis show. I have uninstalled and reinstalled CAD sketcher and watched a video on how to install but still i get no construction axis.

### Addon Version

0.27.2

### Blender Version

3.6.1

### What platform are you running on?

Mac OSX

hlorus, 2023-07-23:
That's a known issue, see the warning in the readme:
Mac OS: Due to an ongoing Blender bug apple Metal users need to change their GPU Backend to OpenGL under: edit->preferences->system->GPU Backend

Closing as duplicate of #375

closed 2023-07-23 by hlorus

IknietBegrijp, 2023-07-25:
thx man