hlorus/CAD_Sketcher #413: Zero Length Vector [BUG]

TimWhiting, 2023-08-31:
### Contact Details

tim.a.whiting@.gmail.com

### Description

I'm setting some properties on my world in a script. Some don't get initialized until after I run the script. When using a driver for a distance constraint, and using one of these computed properties I get the following error in the console: 

```
Python: Traceback (most recent call last):
  File "/Users/timwhiting/Library/Application Support/Blender/3.6/scripts/addons/CAD_Sketcher-main/gizmos/constraint.py", line 210, in draw
    pos = constr.value_placement(context)
  File "/Users/timwhiting/Library/Application Support/Blender/3.6/scripts/addons/CAD_Sketcher-main/model/distance.py", line 397, in value_placement
    coords = self.matrix_basis() @ Vector((outset, offset, 0))
  File "/Users/timwhiting/Library/Application Support/Blender/3.6/scripts/addons/CAD_Sketcher-main/model/distance.py", line 276, in matrix_basis
    angle = v_rotation.angle_signed(x_axis)
ValueError: Vector.angle_signed(other): zero length vectors have no valid angle
```

I was able to fix it temporarily by dragging the points away from the origin, and then refreshing the solver. Essentially all the points collapse into the origin, and then the solver breaks in the code mentioned in the traceback.

### Addon Version

latest

### Blender Version

3.6

### What platform are you running on?

Mac OSX

Andrej730, 2023-09-07:
Can reproduce this:

https://github.com/hlorus/CAD_Sketcher/assets/9417531/a01ad565-0b8e-4d17-a5f1-99cf1f1ce95d

hlorus, 2023-09-07:
Thanks for reporting and reproducing. The commit solves the error from the addon, the solver still doesn't like distance constraints with zero length and will fail in such a scenario. It's best to avoid it by using the coincident constraint.

commit 61c664df (2023-09-07): Constraints: Add angle vector zero length check for distance constraint

Closes #413