hlorus/CAD_Sketcher #427: [BUG] Crash on leaving sketch after converting to mesh

raindropsfromsky, 2023-12-04:
### Contact Details

raindrops.fromsky@gmail.com

### Description

Blender 4.1 crashes when we (a) convert a sketch to mesh and then (b) leave sketch.
(just conversion to mesh or just leaving sketch does not crash Blender.)

This happens regardless of the fact whether we are converting a freshly made sketch or a previously created sketch (by editing it).

If we exit the sketch WITHOUT converting to mesh, Blender does NOT crash.

### Addon Version

0.27.3

### Blender Version

4.1 (actually, BforArtists 4.0.1)

### What platform are you running on?

Windows

hlorus, 2023-12-04:
Could you share the content of the console after the crash?
https://hlorus.github.io/CAD_Sketcher/advanced/#console-output

You'll have to start blender from the console in order to see the console after the crash...

raindropsfromsky, 2023-12-04:
I toggled console with Windows > Toggle system console command.

[image]

Then I added a sketch with a rectangle and a circle, and left sketch without converting them into mesh.
But the console command did not register any activities at all.

Then I edited the sketch, and converted the sketch to mesh.
Again, no change in the log.

As soon as I leave the sketch, BforArtists crashes along with the console.
So I do not see anything captured in it.

After the crash, I found this file:
C:\Users\<userName>\AppData\Local\Temp\CAD_Sketcher-main-Nagesh.log

(unlike the help description, the file name contains "-main-Nagesh" for some reason.)
There is no other log file with CAD_sketcher in its name anywhere on my laptop.
(I checked with **Everywhere**, which indexes all files.)

But it has no content at all. (It has 0 bytes size).
-----------
Finally, I found that I have to [save the file first](https://docs.blender.org/manual/en/latest/troubleshooting/crash.html#crash-log):
So I saved the file first, and then repeated the experiment.
This time it produced a crash report at-
C:\Users\<UserName>\AppData\Local\Temp\<FileName>.crash.txt

This is attached.
[TestingForCrash.crash.txt](https://github.com/hlorus/CAD_Sketcher/files/13544216/TestingForCrash.crash.txt)

hlorus, 2023-12-04:
Die you try the addon with vanilla blender?

To see the console after the crash you can start blender from a console. You should find it by tipping "cmd" in the windows search. From the console you should be enough to just enter "blender".

hlorus, 2023-12-04:
Ahh i'm afraid we cannot take reports into account that apply to alpha versions. If this also happens with stable versions or when 4.1 gets into beta we can have another look.

raindropsfromsky, 2023-12-04:
OK. Thanks!

I tried this with Blender 4.0.2 RC and it does not crash.

Blender 4.1 is scheduled to enter the beta stage on 24th Jan 2024. 
AFAICT, it won't be possible to raise this as a bug for Blender 4.1, right?

hlorus, 2023-12-04:
> AFAICT, it won't be possible to raise this as a bug for Blender 4.1, right? 

How do you mean, report it to the blender Foundation?

raindropsfromsky, 2023-12-05:
Well, when an addon works with Blender 4.0 but not with 4.1, there are two possible reasons:
1. The code of 4.1 is not backward compatible with 4.0
2. Code in 4.1 is broken unintentionally.

In the first case, the addon developers have to know how the code is going to change, so that they can make the addon compatible with the new version. IINW Blender developer forum provides this information in advance to the addon developers, so that they can modify their addons concurrently.

In the second case, the user should be redirected to raise the bug in Blender's GitHub.

As a user, I don't know which case applies here. 
So if you let me know, I can close the bug here, and raise it in the Blender's GitHub.

The final intent is to give a feedback to the right group, so that the correction can be applied at the earliest.

hlorus, 2023-12-05:
It's likely the second scenario as braking changes are usually made in major releases. Reporting the bug upstream makes sense however this usually requires a simplified setup as blender developers probably don't have the time to install addons...

hlorus, 2023-12-05:
Good news, they fixed it already: https://github.com/blender/blender/commit/ce81bdc2840ef96d209c4452e4be95ced5fd7d83

closed 2023-12-05 by hlorus

raindropsfromsky, 2023-12-05:
Great! 
I use development version of bforartists, so this change will hopefully appear in next week's release!

Many thanks!