hlorus/CAD_Sketcher #428: Text not visible

CJagd, 2023-12-06:
Hi

I'm unable to get the text showing 
EG: I have a diameter contraint, but it dimention is not showing up.
[image]

Any Idea ?

RichH53, 2023-12-13:
It does the same for me in Blender 4. Blender 3.62 worked fine with the same file and CAD Sketcher version. The console shows the following error:
.py", line 205, in draw
    blf.size(FONT_ID, text_size, dpi)
TypeError: blf.size() takes exactly 2 arguments (3 given)

closed 2023-12-13 by hlorus

RichH53, 2023-12-13:
> This has been resolved in  <https://github.com/hlorus/CAD_Sketcher/commit/57facf2c91d7f760ad0bf988953961f831ddf5f5> 57facf2
>
> Download the latest version of the addon please.

Thank you. That fixed it.