hlorus/CAD_Sketcher #438: [BUG] constraint icons don't follow the ortho views in C-A-q mode

drewp, 2024-01-11:
### Description

Press C-A-q to go to 4-way view. The constraint icons are placed correctly on the perspective corner view, then those positions are reused on the other 3 views.
[image]

### Addon Version

0.27.3

### Blender Version

4.0.1

### What platform are you running on?

Linux

jos6035, 2026-09-08:
Hi @hlorus, is #438 still needed on currently supported Blender/CAD Sketcher versions, and would you consider assigning it as a paid USD 100 bounty?

Proposed scope: correct constraint-icon positioning in each quad-view region, check perspective and orthographic views, and preserve normal single-view behavior. Preliminary estimate: 3–5 hours, subject to confirming reproduction and the target versions.

The work would be produced with AI assistance, with the technical account and PayPal payout controlled by an adult resident of Peru. Before starting, could you confirm that this is eligible, the acceptance criteria, whether you can pay via PayPal without any upfront costs to us, and when payment would be due after acceptance/merge?

I am seeking a paid assignment and would wait for agreement on scope and payment before starting. Thank you.

knpuma, 2026-09-09:
I traced a likely cause in current main (`c71061a8df8cfa5d10dd76ee975f4331a7149e4d`): `utilities/view.py:get_2d_coords` uses `context.space_data.region_3d`, rather than the current drawing region's `context.region_data`. Both the batched constraint-icon renderer and the picking gizmo call this helper.

I prepared a small local correction and three regression tests using Blender's actual projection utility. Under Blender 5.2.0 LTS, three distinct region projections and the behind-camera case fail before the change and pass afterward; the single-view control passes both ways. Ruff checks pass. These tests use context doubles: visual end-to-end validation and the full extension suite remain outstanding, and a background-mode quad-view check crashed Blender.

Would you consider sponsoring this correction and regression coverage for **US$25**, payable via PayPal after acceptance and merge? I saw the earlier inquiry and am not assuming an assignment or an existing bounty. I can provide the candidate patch and tests for review if this is useful. Prepared with OpenAI Codex assistance, authorized by @knpuma. Payment details would be shared privately.

hlorus, 2026-09-09:
Thanks for looking into this. We're holding off on bounties for now (limited Open Collective funds, and we're focusing maintainer time on stabilizing what's landed), so no paid assignment here.

The diagnosis looks right though: get_2d_coords uses space_data.region_3d instead of context.region_data, which breaks quad view. A normal PR with that fix plus the regression tests is very welcome.

closed 2026-09-28 by hlorus