hlorus/CAD_Sketcher #444: Handling redundant constraints

zevv, 2024-01-29:
Every now and then I run into the problem where the solver tells me I have redundant constraints; no constraints are specifically marked, so the only way to solve this for me is to remove them one by one until I find the offending combination.

This is quite cumbersome, and I'm sure I'm doing something wrong here. What's the proper way to handle this situation?

HummerSee451437, 2024-06-30:
I have the same problem. Did you find a solution by any chance?

zevv, 2024-06-30:
No, I still remove them one by one until I find the offending combination :)

hlorus, 2024-06-30:
We're limited by the capability of the solver which doesn't mark the offending constraints unfortunately. There was a recent addition to avoid creating the same constraints multiple times at least, see:
https://github.com/hlorus/CAD_Sketcher/commit/14762356659dfc39cd9386c2dc4b8db169b06b84

zevv, 2024-06-30:
Is this something we could bring up with the solvespace people? I don't know a single thing about the CAD sketcher implementation, but I guess it would be a sensible feature for a solver to offer this kind of feedback on a failure.

zevv, 2024-06-30:
I just tried Solvespace for the first time, using it is a breeze because I already know how to use CAD sketcher.

However, Solvespace actually is able to indicate the source of redundant constraints, it tells me to "remove any of these to fix this" and shows a list of the constraints causing the conflict. This suggests that the info is available in solvespace itself, so the question would be how to get that out and propagate it to CAD sketcher, right?

In solvespace the case `SolveResult::REDUNDANT_OKAY:` is handled and the list of problematic constraints is dumped:

```
    for(int i = 0; i < g->solved.remove.n; i++) {                          
        hConstraint hc = g->solved.remove[i];                                  
        Constraint *c = SK.constraint.FindByIdNoOops(hc); 
        if(!c) continue;                               
                                                    
        Printf(false, "%Bp   %Fl%Ll%D%f%h%s%E",
            (i & 1) ? 'd' : 'a',
            c->h.v, (&TextWindow::ScreenSelectConstraint),     
            (&TextWindow::ScreenHoverConstraint),
            c->DescriptionString().c_str());                                      
    }                                                                                
```

This suggests the required info should be available at this point, but I'm not sure if this is all available through the Python binding of course.

hlorus, 2024-06-30:
Solvespace unfortunately doesn't ship python bindings. We're using [this](https://pypi.org/project/py-slvs/) python library by realthunder which has some memory improvements but also seems to be pinne to a rather old version of solvespace.
The solver does already provide a list of failed constraints (which CAD Sketcher uses and paints in red). Redundant constraints don't seem to be communicated, maybe it was added in a later version of the solver or is just not exposed in the python bi dings we're using.
@vespakoen was looking into adding official solvespace python bindings a while back, this would obviously be the perfect solution but i guess his efforts never made it into the trunk. Here's some more info on the topic: https://github.com/hlorus/CAD_Sketcher/pull/320

hlorus, 2026-09-04:
This is now handled by the native-curves solver work:

- When a sketch is over-constrained, the solver reports `REDUNDANT_OK` and **marks the constraints in the redundant group as failed**, so they show up red in the constraint list and gizmos — you can see which ones are involved instead of removing them one by one.
- Auto-constraints added while drawing that would be redundant are declined/rolled back, so redundancy no longer accumulates on its own.

Note: with several interchangeable constraints it flags the whole redundant group (any of them could be the one to drop), rather than singling out a single "extra" one.

Closing as resolved — please reopen if you still hit a case where the offending constraints aren't flagged.

closed 2026-09-04 by hlorus