hlorus/CAD_Sketcher #447: [BUG] Generated geometry is incorrect and cannot be properly extruded using the solidify modifier

wipmate, 2024-02-13:
### Description

When creating a geometry using 2D arcs in a specific arrangement, the resulting bezier/mesh has some artifacts which make it difficult to continue working with the generated mesh.

Steps to reproduce:
1. Create new blender project
2. Sketch shapes using 2D arcs and lines (must have some complexity, e.g. as shown in the attached video)
3. Convert the sketch to a mesh
4. Apply the "Solidify" modifier and increase thickness to observe the artifacts

https://github.com/hlorus/CAD_Sketcher/assets/10092394/71886418-26cd-4328-9a53-75fad389443b

Download the model as shown in the video:
[broken_sketch_solidify.zip](https://github.com/hlorus/CAD_Sketcher/files/14271917/broken_sketch_solidify.zip)

(Apparently I don't get any logs from the plugin.)

PS: Huge shoutout to this project, it has already served me well in the past and I hope to continue my work with it!

### Addon Version

0.27.3

### Blender Version

4.0

### What platform are you running on?

Windows

dtremain, 2024-06-25:
This is similar to my issue #440. I have reproduced this bug using points and lines, and changing the Mesh Curve Resolution to 1 did not "fix" it. The .zip contains the .blend file and a .txt file that describes the steps I used to create it. 
[Bug440.zip](https://github.com/user-attachments/files/15963173/Bug440.zip)
[image]

I can create a cross shaft using two sketches - one a horizontal rectangle, the other a vertical rectangle, and then boolean UNION them together.
You MUST use the Point / Line approach if you encase the plus object (shaft) in a circle to get a bore because the two union'ed rectangles do not work.

jmsunseri, 2024-07-29:
i'm experience a similar issue (see sketch) with a bunch of holes in a flat plane .  i tired the "trick" to see if different mesh curve resolutions would solve the issue but so far i'm unable to find one.  the best i can tell it's an issue with face orientation so that the solidify is happening in the opposite direction indicated by the offset 
[image]

[image]

This is a serious blocker. Anything aside from the simplest topology results in issues.  

I seem to be able to manually work around the issue for now by extruding manually

Miguel-Rodrigues, 2024-08-29:
A workaround that I documented on the Discord before is to remesh the object using the mesh to curve and fill curve nodes on geometry nodes.

I suspect that this happens due to the order of how the model was sketched and constrained, causing the creation of excess points with wrong rotations, when filling the shape.

https://discord.com/channels/882152824249667594/1067824923516616805/1073001503519944826

[image]

KoDeltaS, 2024-09-03:
Hi @wipmate , @dtremain , @jmsunseri , @hlorus , I tried to reproduce your issues and found a solution that works and the reason causing the problem. 

The problem comes from the fact that Fill Shape doesn’t work well (it’s probably like a simple F in Edit Mode). 

Solution:
-Update the sketches by unchecking Fill Shape.
-Exit the sketch.
-Duplicate your mesh (if you go back to sketch mode, the changes will be lost).
-Select the copied mesh, in Edit Mode (select everything with A) use ALT + F to fill. If you have complex geometry and ALT + F doesn’t work, create face "islands" manually.

I think it would be advisable to recommend using a step-by-step workflow:

-Sketch for basic geometries
-Sketch of shapes to use as booleans

[image]
[image]
[image]

dtremain, 2024-09-04:
I'm using blender 4.2.1, Windows 10

I created a sketch, added two 4 x 10 rectangles, selected Mesh, turned off fill, renamed object, and tried to Leave the sketch, and got the attached error message.

I tried not renaming the object and got the same error.

David Tremain

F_Bug.png (view on web)<https://github.com/user-attachments/assets/6f91dc2e-942a-48d7-912f-0d485e7e30b6>
Cross_Islands.png (view on web)<https://github.com/user-attachments/assets/9898820e-a4ce-4def-bd86-93abfb5a41bc>
Cross_Z.png (view on web)<https://github.com/user-attachments/assets/bb303ba6-cf29-41b9-849f-370e0d65cfda>

hlorus, 2024-09-04:
@dtremain the attachment seems to be missing, you might have to post it on GitHub directly.

dtremain, 2024-09-04:
[image]

dtremain, 2024-09-04:
The error seems to be related to Blender 4.2.1 LTS. I didn't have a problem in 4.1.

hlorus, 2024-09-05:
Download the latest version of the addon please.

ninazu, 2025-01-30:
[image]

Same issue with simply cross + solidify modifier (also I tried Extrude from Sketcher tools)

Versions: 
- Addon: 0.27.5
- Blender: 4.3.2