hlorus/CAD_Sketcher #452: Support Blender 4.20 and above

hlorus, 2024-03-16:
- [ ] Extension support 
- [ ] Bundle solver module
- [ ] Reduce zip size

bEPHen, 2024-07-18:
I would almost increase this priority due to Blender 4.2 LTS is released.

hlorus, 2024-07-19:
What error did you face? It should already work, the remaining points here are for documentation and improvements for the installation through the new extension system.

henimex, 2024-07-19:
cannot import name 'bl_info' from 'bl_ext.user_default.CAD_Sketcher' (C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher|__init__.py)

hlorus, 2024-07-19:
> Hi im not an old user. just try to use your add on based on a video tutorial. when im trying to enable it im getting this error. i just tried manualy copy files to location.
> 
> cannot import name 'bl_info' from 'bl_ext.user_default.CAD_Sketcher' (C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher|__init__.py)

Where did you download the addon from?

hlorus, 2024-07-19:
>

Where did you download the addon from? Try it with the latest version from github please.

henimex, 2024-07-19:
I directly downloaded from github mate. after your post i deleted everything and redownload... same error apeared. Blender 4.2 is there anything else i could try

hlorus, 2024-07-19:
With 4.2 there's now also the extension system. Did you also try to install it though this?

Bavac, 2024-07-20:
Hi, I tried through the extension system, still get the same error. 

Posted the traceback. From what I can see the problem seems to be that the utilities script can't access the global variable bl_info. If I comment out the check that gets the version number from the variable and instead let it get the version from the manifest. It works fine.  Hope this helps:)

```Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender 4.2\4.2\scripts\modules\addon_utils.py", line 482, in enable
    mod.register()
  File "C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\__init__.py", line 56, in register
    logger.info("Enabled CAD Sketcher base, version: {}".format(get_addon_version()))
                                                                ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\utilities\__init__.py", line 28, in get_addon_version
    version = get_addon_version_tuple()
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\utilities\__init__.py", line 14, in get_addon_version_tuple
    return get_bl_info()["version"]
           ^^^^^^^^^^^^^
  File "C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\utilities\__init__.py", line 6, in get_bl_info
    from .. import bl_info
ImportError: cannot import name 'bl_info' from 'bl_ext.user_default.CAD_Sketcher' (C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\__init__.py)```

closed 2024-07-20 by hlorus