hlorus/CAD_Sketcher #456: [BUG] Module not found error?

liluzibird, 2024-04-03:
### Contact Details

alberth03@protonmail.com

### Description

System environment is Arch Linux (6.7.9-arch1-1). I was trying to add the zipped package to Blender and encountered this error after trying to hit the checkbox:

````
Read prefs: "/home/liluzibird/.config/blender/4.1/config/userpref.blend"
Traceback (most recent call last):
  File "/usr/share/blender/4.1/scripts/modules/addon_utils.py", line 376, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/liluzibird/.config/blender/4.1/scripts/addons/CAD_Sketcher-main/__init__.py", line 6, in <module>
    from .registration import register_base, unregister_base, register_full, unregister_full
  File "/home/liluzibird/.config/blender/4.1/scripts/addons/CAD_Sketcher-main/registration.py", line 1, in <module>
    from .utilities.register import module_register_factory
  File "/home/liluzibird/.config/blender/4.1/scripts/addons/CAD_Sketcher-main/utilities/__init__.py", line 1, in <module>
    import pathlib, toml
ModuleNotFoundError: No module named 'toml'
```

### Addon Version

latest

### Blender Version

4.1.0

### What platform are you running on?

Linux

liluzibird, 2024-04-03:
I did install py_slvs on my system through pip install thinking that would fix this issue, but that didn't help.

colossatr0n, 2024-04-08:
@liluzibird I ran into this same issue and was able to fix it by directly installing the `toml` module using the Python binary that is packaged with the Blender application. I'm on MacOS, so I ran the following:

```sh
/Applications/Blender.app/Contents/Resources/2.81/python/bin/python3.7m -m pip install toml
```

This got rid of the error.

hlorus, 2024-04-08:
The latest commit should resolve this

closed 2024-04-08 by hlorus

trabulium, 2024-04-08:
Just a comment on a similar line. I just downloaded and installed Blender 4.10 on MacOS Ventura 13.5 and the latest Sketcher. Module seemed to enable fine but Menu didn't show. Opening Blender from the terminal did nothing:

running the following:
import bpy
bpy.ops.preferences.addon_enable(module="CAD_Sketcher-main")

Gave me:

CAD_Sketcher-main:{WARNING}: Solvespace module isn't available, only base modules registered
No module named 'py_slvs'

I then ran:

import sys
print(sys.executable)

to give me the full path to the Blender Python binary:

then in my main terminal, I ran:

 /Applications/Blender.app/Contents/Resources/4.1/python/bin/python3.11 -m pip install py_slvs
 
 After that, Sketcher registered fine.