hlorus/CAD_Sketcher #463: Versioning not working since changes to toml versioning were made [BUG]

TimStrauven, 2024-04-21:
### Description

I noticed this when trying to save a blender file with the current versioning system, the following error pops up:
ValueError: bpy_struct: item.attr = val: sequences of dimension 0 should contain 3 items, not 10

The reason this happens is I think that a tuple is expected, however that tuple is converted to string and is then being sent to blender.

The old code before the changes was:
```Python
def get_addon_version():
    from .import bl_info
    return bl_info["version"]

def write_addon_version(context):
    version = get_addon_version()

    logger.debug("Writting addon version: " + str(version))

    # NOTE: Version is written to every scene because there's no global
    # place we can put this value
    for scene in bpy.data.scenes:
        scene.sketcher.version = version
```

the changes splitted the get_addon_version function into another file and made get_addon_version return a string:

```python
def get_bl_info():

    from .. import bl_info
    return bl_info

def get_addon_version() -> str:
    """Return addon version from manifest file"""

    if version < (4, 20):
        return str(get_bl_info()["version"])

    manifest = pathlib.Path(__file__).parent.parent / "blender_manifest.toml"
    try:
        import toml
        return toml.load(manifest)["blender"]
    except Exception:
        return ""
```

The bigger issue is that it seems like any file that has been saved from new since this change with blender version < 4.20 will have no version info at all.... 
So version updating does not seem to happen anymore because it passes here without a set version prop:
```python
    for scene in bpy.data.scenes:
        props = scene.sketcher

        if not props.is_property_set("version"):
            continue
```

### Addon Version

0.27.3

### Blender Version

4.1

### What platform are you running on?

Linux

hlorus, 2024-04-21:
Good catch! Are you planning to submit a PR?

TimStrauven, 2024-04-21:
I don't know the history behind al the versioning changes, so it might be better if you have a look at it.

hlorus, 2024-04-21:
Resolved with:
https://github.com/hlorus/CAD_Sketcher/commit/f2978002a7e69c717869fa4e34632e7ef6232eae

closed 2024-04-21 by hlorus