hlorus/CAD_Sketcher #469: [BUG] Copy-pasting sketches loses some entities

UltraBlackLinux, 2024-07-07:
### Description

When you copy and paste everything in a sketch into another sketch some entities together with their related entities and constraints vanish unpredictably.

In this case it's the circles that vanish

[image]
[image]
(Don't ask me what's going on with the weirdly offset circles in both screenshots. That seems to be a bug too since I never moved them...)

### Addon Version

0.20

### Blender Version

4.1.1

### What platform are you running on?

Linux

hlorus, 2024-07-19:
How did you select the entities? Box select has the limitation that it cannot select entities that are overlapped by another entity. If you use the select all operator it should work...

UltraBlackLinux, 2024-07-19:
it was box select, and the vertices were attached to one another to form a single vertex, which I'd expect to not break like that.

hlorus, 2024-07-19:
It's a known limitation of how selection works. [This](https://github.com/hlorus/CAD_Sketcher/issues/50) is the relevant issue.
You'll have to use select_all (ctrl+A) to ensure that no entities are left out.