hlorus/CAD_Sketcher #470: [BUG] modifiers on generated objects not preserved

UltraBlackLinux, 2024-07-07:
### Description

When leaving a sketch and having it filled with a mesh, all modifiers and position changes made to the "rendered" object disappear. 

since you pretty much have to rely on modifiers and positioning to handle boolean operations and even set an object thickness, having those reset is a huge pain as you have to dial them in again and again after making even a simple change.

### Addon Version

0.20

### Blender Version

4.1.1

### What platform are you running on?

Linux

hlorus, 2024-07-20:
I cannot reproduce this issue, can you describe exact steps to reproduce please?

UltraBlackLinux, 2024-07-21:
1. make a sketch
2. fill it with a solid
3. add a modifier to that solid
4. edit and leave the sketch
5. modifiers (and more?) gone

hlorus, 2024-07-21:
Sorry, i still have a few questions:
- what convert type are you using, mesh or curve?
- what do you mean with nr.2, are you using a modifier to fill an outline? There's a setting "Fill Shape" right next to the convert type, can't you use that?

hlorus, 2024-07-21:
I just saw that you're using a rather old version of the addon (0.20). Could you try - with the latest version from github?

UltraBlackLinux, 2024-07-22:
hmm well I can't get the latest git version to work at all:
[image]

hlorus, 2024-07-22:
That's seemingly not the latest version but, there have been some commits in the last 3 days...

hlorus, 2024-08-21:
Closing this as it's likely outdated, feel free to reopen if the problem persists.

closed 2024-08-21 by hlorus