hlorus/CAD_Sketcher #474: [BUG] Error trying to convert sketch to mesh

rmastrobiso, 2024-08-05:
### Contact Details

mastrobiso@gmail.com

### Description

When I hit "Leave: Sketch" after selecting the "Mesh" value in the "Convert to Type" dropdown, this happens:

Python: Traceback (most recent call last):
  File "C:\zaq\Blender\4.2\scripts\addons\CAD_Sketcher-main\operators\set_sketch.py", line 20, in execute
    return activate_sketch(context, self.index, self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\zaq\AppData\Roaming\Blender Foundation\Blender\4.2\scripts\addons\CAD_Sketcher-main\operators\utilities.py", line 186, in activate_sketch
    update_convertor_geometry(context.scene, sketch=last_sketch)
  File "C:\zaq\AppData\Roaming\Blender Foundation\Blender\4.2\scripts\addons\CAD_Sketcher-main\converters.py", line 135, in update_convertor_geometry
    sketch.target_curve_object = object
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\zaq\AppData\Roaming\Blender Foundation\Blender\4.2\scripts\addons\CAD_Sketcher-main\utilities\bpy.py", line 85, in unique_attribute_setter
    self[name] = value
    ~~~~^^^^^^
TypeError: Cannot assign a 'Object' value to the existing 'target_curve_object' Group IDProperty

### Addon Version

0.27.3

### Blender Version

4.2

### What platform are you running on?

Windows

hlorus, 2024-08-06:
You'll have to get the latest version from github to use it with blender 4.2.

closed 2024-08-06 by hlorus

Buddy89876, 2024-08-12:
Hello hlorus,

Is the latest version 0.27.3? That's the version I am on, as well as blender version 4.2, and I am experiencing the same problem.

hlorus, 2024-08-13:
No, it's 27.5. You can grab it from github.

void-dragon, 2024-08-18:
I have version sketcher 0.27.5 and blender 4.2 on mac os and experience this bug, right now !

hlorus, 2024-08-19:
> I have version sketcher 0.27.5 and blender 4.2 on mac os and experience this bug, right now !

What output are you getting in the console?