hlorus/CAD_Sketcher #479: [BUG] Constraints Angle supplementary and Distance switch from Horizontal to Vertical or vice versa edits the sketch badly

dimitarsp, 2024-08-28:
### Description

When creating an angle constraint, and clicking on measuring supplementary angle, it tries to adjust the sketch with the current angle applied in the supplementary direction.

For exmample if I am measuring a 30° ∢, I expect on clicking the supplementary option to give me 70°. Instead, it adjusts the sketch so that the 70° ∢ becomes 30° 

Similarly, when creating a distance constraint, swtiching from None to Vertical or Horizontal, it keeps the non-aligned distance and applies it to either the vertical or horizontal direction.

For example if I am measuring a line to be 8m at a 45° ∢, and choose instead Vertical, it applies that 8m to the vertical so that the segment becomes 11.314m long

This behaviours seems to be this way only if adjsuting the options in the previous command area in bottom left-hand corner. When adjusting and the options of constraings via right-clicking on the constraints, it works as expected

### Addon Version

0.27.5

### Blender Version

2.42.1

### What platform are you running on?

Windows

MagicBOTAlex, 2024-09-01:
Here's some images of what they mean:

#### Original example
[image]

#### Current
[image]

#### What they want
[image]

###### Correct me if I'm wrong

hlorus, 2024-09-02:
IMO the behavior should not always be the same, ideally it should depend on the order the values are set. 
-> If a distance constraint is added and the distance value is set before the alignment the value shouldn't be changed, even if that edits the sketch.
-> If the order is the opposite the value should be edited but the placement of entities shouldn't be changed.