hlorus/CAD_Sketcher #480: [BUG] Copy-Paste and duplicate not working

Polly-cracker, 2024-08-29:
### Contact Details

polletje@gmail.com

### Description

Pasting objects or using duplicate gives an error and nothing is pasted or duplicated, 
See error trace below.

Python: Traceback (most recent call last):
  File "C:\Users\<user>\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\operators\copy_paste.py", line 106, in execute
    paste(context, buffer)
  File "C:\Users\<user>\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\serialize.py", line 161, in paste
    scene_from_dict(scene, final_dict)
  File "C:\Users\<user>\AppData\Roaming\Blender Foundation\Blender\4.2\extensions\user_default\CAD_Sketcher\serialize.py", line 37, in scene_from_dict
    scene["sketcher"].update(original)
TypeError: Cannot assign a 'dict' value to the existing 'entities' Group IDProperty

### Addon Version

0.27.5

### Blender Version

4.2.1 LTS

### What platform are you running on?

Windows

hlorus, 2024-08-29:
This is fixed since https://github.com/hlorus/CAD_Sketcher/commit/6cabd1fb3f5635df569185d151ae7a2790653028

You can download the latest version from github.

closed 2024-08-29 by hlorus

Polly-cracker, 2024-08-29:
thx for you quick reply hlorus and your amazing work!
I downloaded CAD Sketcher from gumroad, so that's not the latest I should use?
Thanks again and keep the good work!!