hlorus/CAD_Sketcher #482: [Feature] Icons for tools

MagicBOTAlex, 2024-09-01:
## Motivation
I have noticed a lot of missing icons or duplicated icons for a lot of tools. Here's some examples:
[image]
[image]
[image]
[image]

## Description
I want some icons for quicker differentiation of different tools.

## Mockup
Distance
[image]
Add 2D line:
[image]
Add 3D line:
[image]

### Here's a checklist of icons:
- [ ] "Add 2D point" or "Add 3D point"
- [ ] "Add 2D line" or "Add 3D line"
- [ ] "Distance"
- [ ] "Diameter"
- [ ] "Angle"
- [ ] "Construction Mode" the checkbox and if possible, the selection menu
###### Please add more if I missed some

hlorus, 2024-09-03:
Here's the file used to create the icons for the workspacetools: [icon_geom_BCS.zip](https://github.com/user-attachments/files/16845295/icon_geom_BCS.zip)

MagicBOTAlex, 2024-09-03:
Whaaat? It's made in Blender!
I didn't expect the icons to be made in Blender. Crazy