hlorus/CAD_Sketcher #483: [BUG] - Add a Sketch Impossible to Select the workplane

IGLOU-EU, 2024-09-01:
### Description

**Description:**
When I select "Add a Sketch," the planes are displayed, but it is impossible to select one. There is no hover effect, and clicking to validate cancels the creation. This issue is new, and I did not have this problem the last time, even though it is the same version of Blender. I have tested under Blender 4.2.1, 4.2.0, 4.1.1, and 4.1.0, and the same issue occurs.

**Steps to Reproduce:**
1. Open Blender.
2. Select "Add a Sketch."
3. Observe the displayed planes.
4. Attempt to select a plane.

**Expected Result:**
- A plane should be selectable, with a hover effect indicating the selection.
- Clicking should validate the selection and not cancel the creation.

**Actual Result:**
- No plane is selectable.
- No hover effect is present.
- Clicking cancels the creation.

**System**
- Linux 6.10.7-arch1-1 x86_64 GNU/Linux
- Python 3.12.5
- pip 24.2 from /usr/lib/python3.12/site-packages/pip (python 3.12)

```
Name: py_slvs
Version: 1.0.6
Summary: Python binding of SOLVESPACE geometry constraint solver
Home-page: https://github.com/realthunder/slvs_py
Author: Zheng, Lei
Author-email: realthunder.dev@gmail.com
License: Gnu General Public License 3.0
Location: /home/adrien/.local/lib/python3.12/site-packages
```

**Debug Log:**
```
bl_ext.user_default.CAD_Sketcher:{INFO}: Logging into: /tmp/bl_ext.user_default.CAD_Sketcher-adrien.log
Error: Unable to retrieve addon version
bl_ext.user_default.CAD_Sketcher:{INFO}: Enabled CAD Sketcher base, version: 0.0.0
bl_ext.user_default.CAD_Sketcher.handlers:{DEBUG}: Append <version_update> builtin handlers:
  - do_versioning
bl_ext.user_default.CAD_Sketcher.handlers:{DEBUG}: Append <save_pre> builtin handlers:
  - write_addon_version
bl_ext.user_default.CAD_Sketcher:{INFO}: Solvespace available, fully registered modules
bl_ext.user_default.CAD_Sketcher.versioning:{DEBUG}: Check versioning
Error: Unable to retrieve addon version
bl_ext.user_default.CAD_Sketcher.versioning:{WARNING}: Scene Scene was saved with a newer version of the addon
bl_ext.user_default.CAD_Sketcher.versioning:{DEBUG}:
Update existing file to version (0, 0, 0)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
  - SlvsNormal3D(1)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
 - SlvsNormal3D(1)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
  - SlvsNormal3D(1)
  - SlvsNormal3D(2)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
 - SlvsNormal3D(1)
 - SlvsNormal3D(2)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
  - SlvsNormal3D(1)
  - SlvsNormal3D(2)
  - SlvsWorkplane(0)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
 - SlvsNormal3D(1)
 - SlvsNormal3D(2)
 - SlvsWorkplane(0)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
  - SlvsNormal3D(1)
  - SlvsNormal3D(2)
  - SlvsWorkplane(0)
  - SlvsWorkplane(1)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
 - SlvsNormal3D(1)
 - SlvsNormal3D(2)
 - SlvsWorkplane(0)
 - SlvsWorkplane(1)
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: --- Start solving ---
All:False, Sketch:None, g:2
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize entities:
  - SlvsPoint3D(0)
  - SlvsNormal3D(0)
  - SlvsNormal3D(1)
  - SlvsNormal3D(2)
  - SlvsWorkplane(0)
  - SlvsWorkplane(1)
  - SlvsWorkplane(2)
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Initialize constraints:
bl_ext.user_default.CAD_Sketcher.solver:{INFO}: Successfully solved sketch.
bl_ext.user_default.CAD_Sketcher.solver:{DEBUG}: Update entities from solver:
 - SlvsPoint3D(0)
 - SlvsNormal3D(0)
 - SlvsNormal3D(1)
 - SlvsNormal3D(2)
 - SlvsWorkplane(0)
 - SlvsWorkplane(1)
 - SlvsWorkplane(2)
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches
```

[image]

### Addon Version

latest

### Blender Version

4.2.1 LTS

### What platform are you running on?

Linux

hlorus, 2024-09-02:
Just to be clear, workplanes are only selectable on their edge, not on their face. Also one of the addon's workspace tools (usually the addon's select tool) has to be active.

Could you try to run the "Write Selection Texture" operator from the debug panel when the three base workplanes are visible? This will create a new image data block in blender. There you should see the outlines of the workplanes.

IGLOU-EU, 2024-09-02:
Same result when I run "Write Selection Texture", with this console output
```
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
bl_ext.user_default.CAD_Sketcher.draw_handler:{DEBUG}: Update geometry batches:
```

But at start, we can see `Error: Unable to retrieve addon version`, this is odd no ?

IGLOU-EU, 2024-09-02:
Hum, I have a Traceback when a quit Blender
```
Traceback (most recent call last):
  File "/home/adrien/.config/blender/4.2/extensions/user_default/CAD_Sketcher/utilities/register.py", line 43, in unregister
    m.unregister()
  File "/home/adrien/.config/blender/4.2/extensions/user_default/CAD_Sketcher/keymaps.py", line 394, in unregister
    km.keymap_items.remove(kmi)
RuntimeError: Error: KeyMapItem 'WM_OT_tool_set_by_id' not found in KeyMap 'Object Mode'

bl_ext.user_default.CAD_Sketcher.handlers:{DEBUG}: Remove <version_update> builtin handlers:
  - do_versioning
bl_ext.user_default.CAD_Sketcher.handlers:{DEBUG}: Remove <save_pre> builtin handlers:
  - write_addon_version

Blender quit
```

Re-installing the solver don't do anything
```
$ pip uninstall --break-system-packages py-slvs
Found existing installation: py_slvs 1.0.6
Uninstalling py_slvs-1.0.6:
  Would remove:
    /home/adrien/.local/lib/python3.12/site-packages/py_slvs-1.0.6.dist-info/*
    /home/adrien/.local/lib/python3.12/site-packages/py_slvs/*
Proceed (Y/n)?
  Successfully uninstalled py_slvs-1.0.6
```
```
$ pip install --break-system-packages --no-cache-dir py-slvs
Defaulting to user installation because normal site-packages is not writeable
Collecting py-slvs
  Downloading py_slvs-1.0.6.tar.gz (2.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.4/2.4 MB 107.1 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: py-slvs
  Building wheel for py-slvs (pyproject.toml) ... done
  Created wheel for py-slvs: filename=py_slvs-1.0.6-cp312-cp312-linux_x86_64.whl size=285493 sha256=419c4384c6a1d7061c2b9e08d49ea83a4c8534e4e9435f02672d0645b72a4d56
  Stored in directory: /tmp/pip-ephem-wheel-cache-o_h85isq/wheels/42/e5/fe/da622698b5e3fa04e5f0bb1914de04435b12cb430761b3d2ff
Successfully built py-slvs
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.6
```

hlorus, 2024-09-03:
> And at start, we can see `Error: Unable to retrieve addon version`, this is odd no ?

This was a bug but should already be resolved with the newest version from github, could you try with that one? Gumroad might be missing that fix.

Not sure about the keymap issue tho...

lividhen, 2024-09-08:
Same issue here. Tried with commit 3338ad1.
Linux 6.10.7-zen1-1-zen (arch)
Python 3.12.5

IGLOU-EU, 2024-09-09:
I have tested with the main, from the first September. And just now from 3338ad1a922c1d4e5f8ab7f0bc1e90f1d0f596ba :smile:

heyfixit, 2024-09-11:
I've just now tested with github zip and gumroad, same result for me too.
OS: ArcoLinux 
Kernel: 6.10.8-arch1-1 
WM: herbstluftwm

Blender 4.2.1 LTS

I've tried both the flatpak and pacman, and have done complete wipes of all blender related files in between, still can't select a plan no matter how I'm rotated or where the mouse is.

heyfixit, 2024-09-11:
Went back to here on a whim: [d71f674](https://github.com/hlorus/CAD_Sketcher/commit/d71f674f9aa15601efd6c0542f749a083a16682b)

No dice.

hlorus, 2024-09-11:
Maybe someone could try uninstalling the solver and addon and reinstalling the addon through the new extension mechanism which is described here:
https://hlorus.github.io/CAD_Sketcher/installation/

That way the correct solver module is shipped with the addon.

IGLOU-EU, 2024-09-11:
@hlorus Same :grimacing: 

Steps :
- Disable addon on Blender 4.2
- Remove addon directory ~/.config/blender/4.2/extensions/user_default/CAD_Sketcher
- Remove installed solver with pip + remove installed solver from Blender 4.2 ~/.config/blender/4.2/extensions/.local/lib/*
- Quit blender
- Clone repos inside ~/.config/blender/4.2/extensions/user_default/CAD_Sketcher
- Start blender and enable CADS
- Not working :(

[image]

lividhen, 2024-09-11:
If someone has some free time could you check to see if its working on any other kernels or distros?

IGLOU-EU, 2024-09-11:
> If someone has some free time could you check to see if its working on any other kernels or distros?

Not this week for me, sorry :disappointed:

Shaddy-ElDesouky, 2024-09-12:
It's working for me, I only added the addon yesterday:
OS: MX Linux 23.3
Blender: 4.2.0
Kernel: 6.10.7-1-liquorix-amd64 [6.10-7~mx23ahs]
Python 3.11.2
I couldn't find the extension on https://extensions.blender.org/search/?q=cad nor by searching from inside Blender so I downloaded the addon from github and chose install from disk (just like old days).

lividhen, 2024-09-12:
I will see if I can try 6.10.7.
Is anyone else besides shaddy using an nvidia GPU?

heyfixit, 2024-09-12:
`echo -e "GPU: $(lspci | grep -i vga | cut -d ':' -f3 | xargs)\nDisplay: ${XDG_SESSION_TYPE:-Unknown}\nKernel: $(uname -r)"`

```
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] (rev c1)
Display: x11
Kernel: 6.10.8-arch1-1
```

heyfixit, 2024-09-12:
Worked as expected on an ancient thinkpad:
```
GPU: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
Display: x11
Kernel: 6.10.6-zen1-1-zen
```

Same distro as my AMD machine so can't blame herbstluftwm or any of my window manager funkiness.
Will leave off the nVidia test since we see it working there for everyone.

I took the common path for installation as directed on cad sketcher site, everything went fine.

IGLOU-EU, 2024-09-12:
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] (rev d1)
Display: wayland 1.23.0-1
Kernel: 6.10.7-arch1-1
Python: 3.12.5
CADS: Main

Shaddy-ElDesouky, 2024-09-12:
[CODE]Graphics:
  Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] vendor: Dell driver: i915 v: kernel
    arch: Gen-9.5 process: Intel 14nm built: 2016-20 ports: active: eDP-1 empty: DP-1, DP-2, DP-3,
    HDMI-A-1, HDMI-A-2 bus-ID: 00:02.0 chip-ID: 8086:3e9b class-ID: 0300
  Device-2: NVIDIA TU106GLM [Quadro RTX 3000 Mobile / Max-Q] vendor: Dell driver: nvidia
    v: 535.183.06 non-free: 530.xx+ status: current (as of 2023-03) arch: Turing code: TUxxx
    process: TSMC 12nm FF built: 2018-22 pcie: gen: 1 speed: 2.5 GT/s lanes: 16 link-max: gen: 3
    speed: 8 GT/s bus-ID: 01:00.0 chip-ID: 10de:1f36 class-ID: 0300
  Device-3: Realtek Integrated_Webcam_HD type: USB driver: uvcvideo bus-ID: 1-11:5
    chip-ID: 0bda:5532 class-ID: 0e02 serial: <filter>
  Display: x11 server: X.Org v: 1.21.1.7 compositor: xfwm v: 4.18.0 driver: X:
    loaded: modesetting,nvidia unloaded: fbdev,nouveau,vesa alternate: nv dri: iris gpu: i915
    display-ID: :0.0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96 s-size: 508x285mm (20.00x11.22") s-diag: 582mm (22.93")
  Monitor-1: eDP-1 model: AU Optronics 0x139d built: 2019 res: 1920x1080 hz: 60 dpi: 128
    gamma: 1.2 size: 381x214mm (15x8.43") diag: 437mm (17.2") ratio: 16:9 modes: 1920x1080
  API: OpenGL v: 4.6 Mesa 23.1.2-1~mx23ahs renderer: Mesa Intel UHD Graphics 630 (CFL GT2)
    direct-render: Yes[/CODE]

hlorus, 2024-09-12:
> Went back to here on a whim: [d71f674](https://github.com/hlorus/CAD_Sketcher/commit/d71f674f9aa15601efd6c0542f749a083a16682b)

This was the latest commit that touched the drawing/selection stuff so it would be great to try the addon versio before the change which would be: 6cabd1fb3f5635df569185d151ae7a2790653028

heyfixit, 2024-09-12:
I've tried a few distros/kernels and it worked on these two:

1) Ubuntu 20.04:
```
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Device 73bf (rev c1)
Display: x11
Kernel: 5.15.0-67-generic
```

2) The most recent Ubuntu LTS:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] (rev c1)
Display: x11
Kernel: 6.8.0-41-generic
```

With my current Arch install, neither of these kernels worked:
1) linux-zen 6.10.9.zen1-2
2) linux-lts 6.6.50-2

zevv, 2024-09-12:
Same problem here.  I have not updated blender or CAD sketcher since the problem started to occur, I left things in working order last week. I have updated some debian packages in the mean time though.

```
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
```

```
Linux version 6.8.11-amd64 (debian-kernel@lists.debian.org) (x86_64-linux-gnu-gcc-13 (Debian 13.2.0-25) 13.2.0, GNU ld (GNU Binutils for Debian) 2.42) #1 SMP PREEMPT_DYNAMIC Debian 6.8.11-1 (2024-05-26)
```

```
Blender 4.0.2
	build date: 2023-12-05
	build time: 08:48:50
	build commit date: 2023-12-05
	build commit time: 07:41
	build hash: 9be62e85b727
	build platform: Linux
	build type: release
	build c flags:  -Wall -Werror=implicit-function-declaration -Wstrict-prototypes -Werror=return-type -Werror=vla -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wformat-signedness -Wrestrict -Wno-stringop-overread -Wno-stringop-overflow -Wnonnull -Wabsolute-value -Wuninitialized -Wredundant-decls -Wshadow -Wimplicit-fallthrough=5 -Wno-error=unused-but-set-variable  -std=gnu11 -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
	build c++ flags:  -Wuninitialized -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Werror=return-type -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wundef -Wcomma-subscript -Wformat-signedness -Wrestrict -Wno-suggest-override -Wuninitialized -Wno-stringop-overread -Wno-stringop-overflow -Wimplicit-fallthrough=5 -Wundef -Wmissing-declarations  -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
	build link flags:  -Wl,--version-script='/home/blender/git/blender-v400/blender.git/source/creator/symbols_unix.map'
	build system: CMake
```

```
0b:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c7)
```

heyfixit, 2024-09-12:
> This was the latest commit that touched the drawing/selection stuff so it would be great to try the addon versio before the change which would be: [...]

Just tried it, no dice.

The LLMs are thinking it`s coming down to:
1) Kernel modules/firmware
2) Graphics stack configs.
3) System configs.

I might attempt to pick the needle out of that haystack, it's probably gonna involve a lot of hallucinating between me and my LLM friends tho.

lividhen, 2024-09-12:
> Same problem here. I have not updated blender or CAD sketcher since the problem started to occur, I left things in working order last week. I have updated some debian packages in the mean time though.
> 
> Debug output shows `CAD_Sketcher-main.draw_handler:{DEBUG}: Update geometry batches:` when entering and leaving the workplane area with the mouse, but not selectable indeed.
> 
> ```
> No LSB modules are available.
> Distributor ID:	Debian
> Description:	Debian GNU/Linux trixie/sid
> Release:	n/a
> Codename:	trixie
> ```
> 
> ```
> Linux version 6.8.11-amd64 (debian-kernel@lists.debian.org) (x86_64-linux-gnu-gcc-13 (Debian 13.2.0-25) 13.2.0, GNU ld (GNU Binutils for Debian) 2.42) #1 SMP PREEMPT_DYNAMIC Debian 6.8.11-1 (2024-05-26)
> ```
> 
> ```
> Blender 4.0.2
> 	build date: 2023-12-05
> 	build time: 08:48:50
> 	build commit date: 2023-12-05
> 	build commit time: 07:41
> 	build hash: 9be62e85b727
> 	build platform: Linux
> 	build type: release
> 	build c flags:  -Wall -Werror=implicit-function-declaration -Wstrict-prototypes -Werror=return-type -Werror=vla -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wformat-signedness -Wrestrict -Wno-stringop-overread -Wno-stringop-overflow -Wnonnull -Wabsolute-value -Wuninitialized -Wredundant-decls -Wshadow -Wimplicit-fallthrough=5 -Wno-error=unused-but-set-variable  -std=gnu11 -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
> 	build c++ flags:  -Wuninitialized -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Werror=return-type -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wundef -Wcomma-subscript -Wformat-signedness -Wrestrict -Wno-suggest-override -Wuninitialized -Wno-stringop-overread -Wno-stringop-overflow -Wimplicit-fallthrough=5 -Wundef -Wmissing-declarations  -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
> 	build link flags:  -Wl,--version-script='/home/blender/git/blender-v400/blender.git/source/creator/symbols_unix.map'
> 	build system: CMake
> ```
> 
> ```
> 0b:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c7)
> ```

X11 or wayland on your arch install? Seems like the issue might be wayland.

zevv, 2024-09-12:
> Just tried it, no dice.
>
> The LLMs are thinking it`s coming down to:
>
>     1. Kernel modules/firmware
>
>     2. Graphics stack configs.
>
>     3. System configs.
>
>
> I might attempt to pick the needle out [...]

Not likely the kernel: my uptime is 34 days, and I have definitely been able to use CAD sketcher last week.

zevv, 2024-09-12:
> X11 or wayland on your arch install? Seems like the issue might be wayland.

Sorry, forgot to mention: X11.

My house is proudly wayland-free!

zevv, 2024-09-12:
Interesting fact: I also did not restart my X11 over the last month, still stuff stopped working in the last week.

SleepingCrows, 2024-10-01:
> > X11 or wayland on your arch install? Seems like the issue might be wayland.
> 
> Sorry, forgot to mention: X11.
> 
> My house is proudly wayland-free!

Definitely not a Wayland issue, Tried with both and got the same result. 

```
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6950 XT] (rev c0)
Display: wayland
Kernel: 6.10.11-x64v2-xanmod1-1
```

IGLOU-EU, 2024-10-01:
A new real headache issue ... :slightly_frowning_face:

zevv, 2024-10-01:
It seems that the issue is not related to the workplane in particular, but to any kind of selection of objects. Reproduction:

- Open an existing blender file containing a sketch (a project I made a few weeks ago)
- Select and activate a sketch
- Try to select any object in the sketch

Interesting fact: measurements (red) are selectable, but objects (black) are not. When trying to select an object, I get the following error message:

```Traceback (most recent call last):
  File "/home/ico/.config/blender/4.0/scripts/addons/CAD_Sketcher-main/utilities/highlighting.py", line 81, in invoke
    return self.execute(context)
  File "/home/ico/.config/blender/4.0/scripts/addons/CAD_Sketcher-main/operators/select.py", line 49, in execute
    entity.selected = value
AttributeError: 'NoneType' object has no attribute 'selected'
Error: Python: Traceback (most recent call last):
  File "/home/ico/.config/blender/4.0/scripts/addons/CAD_Sketcher-main/utilities/highlighting.py", line 81, in invoke
    return self.execute(context)
  File "/home/ico/.config/blender/4.0/scripts/addons/CAD_Sketcher-main/operators/select.py", line 49, in execute
    entity.selected = value
AttributeError: 'NoneType' object has no attribute 'selected'
```

Ragebone, 2024-10-27:
I am affected by this as well and i can confirm that the issue is around selecting objects like lines and points.
No issues selecting constraints.

redhotvengeance, 2024-11-02:
I'll also chime in -- just found this add-on and was excited to dive into it but this issue is blocking me. I've tried installing the `py-slvs` several different ways, as well as rolling back the this repo to one of the versions mentioned above. Nothing is working thus far for me.

Blender 4.2.3 LTS
OS: NixOS
Kernel: 6.11.3
WM: Hyprland

hlorus, 2024-11-03:
> I'll also chime in -- just found this add-on and was excited to dive into it but this issue is blocking me. I've tried installing the `py-slvs` several different ways, as well as rolling back the [...]

With blemder 4.2 and above you don't need to manually install the solver module, just install the addon through the new extensions system as described in the addons installation guide.

redhotvengeance, 2024-11-03:
> With blemder 4.2 and above you don't need to manually install the solver module, just install the addon through the new extensions system as described in the addons installation guide.

That was my first approach to using the add-on. It appeared that it found `py-slvs` correctly, but this bug was happening. That's when I switched to several different ways to install the add-on and the dependency, none of which have fixed or worked around the bug.

zevv, 2024-11-03:
CAD sketcher implements two functions to convert between object entity indices and RGB colors. These RGB colors are not visible to the user and only exist in off-screen buffers, but these are used to implement object picking and selecting.

The functions that convert between index and color require the read-back color to be absolutely precise. However, the RGB colors are floating point numbers that get mangled somewhere in the GPU pipeline on some configurations (I'm not sure about the underlying cause though), and the conversion would trip over these numbers when they are slightly off.

I just submitted a small PR that implements a more robust conversion that should be able to better handle rounding errors, this solves the issue for me.

Back to sketching, finally!

PR #494 (2024-11-04): More robust conversion between indices and floating point RGB colors
This should now properly handle rounding and precision errors in the RGB values without mangling the indices, fixes #483
commit de8da47e (2024-11-04): More robust conversion between indices and floating point RGB colors, this should now properly handle rounding and precision errors in the RGB values without mangling the indices, fixes #483 (#494)
commit 5107f408 (2025-02-18): More robust conversion between indices and floating point RGB colors, this should now properly handle rounding and precision errors in the RGB values without mangling the indices, fixes #483 (#494)
closed 2024-11-04 by hlorus