hlorus/CAD_Sketcher #485: [BUG]

Shaddy-ElDesouky, 2024-09-12:
### Contact Details

shadyeldesouky@yahoo.com

### Description

A bug happened! when converting to mesh the output isn't what I expected

### Addon Version

0.27.5

### Blender Version

4.2.0

### What platform are you running on?

Linux

Shaddy-ElDesouky, 2024-09-12:
[image]
the sketch
[image]
the converted mesh in edit mode box selection with x-ray off

what I expected is to:
- have the short line end vertex at the beginning vertex of the next line.    
- have vertices merged automaticaly.

IGLOU-EU, 2024-09-12:
From my understanding of this software, you can only make one figure at the time with CADS, without the choice of number of planes faces. Like if every 2D line were a wall.

Maybe Screw Modifier can be a better solution ?

Shaddy-ElDesouky, 2024-09-12:
@IGLOU-EU I think screw will help in 3D, but I'm still 2D, before CAD-Sketcher the workflow would be drawing in other app like librecad, inkscape, solvespace, FreeCAD ....etc and import drawing in blender as svg or dxf then converting to mesh and fill faces then going 3D with extrude or solidify or screw or what ever, my issue here is the sketch being translated badly on the 2d level, if I have two vertices constrained with coincident they should not break and move when converted.

Shaddy-ElDesouky, 2024-09-12:
OK I found some explanation here https://hlorus.github.io/CAD_Sketcher/integration/ , 

I understand that I should follow the first of the Best practices, could someone explain how to connect entities by sharing a start-/endpoint ?
Best Practices

To avoid running into such limitations try to follow these practices:

    Connect entities by sharing a start-/endpoint, avoid using the coincident constraint to connect entities that are part of a path that you intend to later convert.