hlorus/CAD_Sketcher #487: On Mac M2 it won't start any of the function that generates 2d shapes and gives me this error when I create a circle

Jasssbo, 2024-09-16:
This is the error it gives me:
Python: Traceback (most recent call last):
  File "/Users/andreamazzurana/Library/Application Support/Blender/4.1/scripts/addons/CAD_Sketcher-main/stateful_operator/logic.py", line 512, in modal
    return self.evaluate_state(context, event, event_triggered)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/andreamazzurana/Library/Application Support/Blender/4.1/scripts/addons/CAD_Sketcher-main/stateful_operator/logic.py", line 553, in evaluate_state
    values = to_list(self._get_state_values(context, state, coords))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/andreamazzurana/Library/Application Support/Blender/4.1/scripts/addons/CAD_Sketcher-main/stateful_operator/logic.py", line 519, in _get_state_values
    pos_val = position_cb(context, coords)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/andreamazzurana/Library/Application Support/Blender/4.1/scripts/addons/CAD_Sketcher-main/operators/add_circle.py", line 57, in get_radius
    delta = Vector(pos) - self.ct.co
            ^^^^^^^^^^^
TypeError: mathutils.Vector()

hlorus, 2024-09-24:
What's the addon version you're using?

hlorus, 2026-08-18:
This was reported against the old add-on (Blender 4.1, v0.27.x). The crash was in add_circle.py::get_radius, and that code has since been fully rewritten for the 0.3 native-curves rework, so this specific traceback no longer exists.

Closing as resolved. If you still hit a crash creating shapes on 0.3, please reopen with the new traceback and we will investigate.

closed 2026-08-18 by hlorus