hlorus/CAD_Sketcher #489: [BUG] Extra edges and vertices when converting to filled mesh

dengste, 2024-10-04:
### Contact Details

deng@randomsample.de

### Description

Steps to reproduce:

* Create the following simple sketch:

[image]

The exact dimensions do not matter, but make sure to have an equal constraint at the left and right vertical lines I marked with "A" and "B". Then set "convert type" to "mesh" and select "Fill Shape". This is the result:

[image]

Note the additional horizontal edge and vertices I marked with the red circle. This should not be there and creates problems with further work (like when applying the screw modifier, this creates additional faces I don't want).

Note this only happens when converting to a filled mesh. If the mesh is not filled, it is fine. Likewise, a Bezier will work fine. Also, it only happens when the lines A and B are exactly equal, not matter how this is achieved. For instance, you can remove the equal constraint and use two distance constraints with the same length, it'll also happen, but just the slightest difference in distance will make this problem disappear.

Fortunately, someone has posted a workaround in Discord: you can disable "Fill Shape" and use a geometry node modifier like this to fill the mesh instead:

[image]

Naturally, I'm wondering if maybe this can be implemented in CADS itself?

### Addon Version

latest

### Blender Version

4.2.2

### What platform are you running on?

Linux

dengste, 2024-10-19:
Just saw issue #447 and I guess this is most probably a duplicate of that one, so feel free to close this. I think though the above simple example might be helpful in reproducing/testing this issue.

Chrissie-AI, 2024-12-23:
Gents, recent Blender/ CAD Sketcher user here. Blender 4.2.3 and CAD Sketcher 0.27.5, on macOS 15.1.1.

I've encountered the exakt same behaviour in one of my 1st attempts, including surplus vertices and surplus horizontal line, which I managed to get rid of. Here's how:

It struck me, that the surplus vertices are _unevenly_ spaced, but apparently not randomly. The spacing very much resembles the **_projected_** vertices of an evenly spaced arc/semicircle, like this:
[image]

Zooming in to the left part of the red ellipse shows what I mean:
[image]

So there seems to be a **"stray" arc** being converted, where there should be none. That's the first observation. Note how the number of segments of this stray arc (and hence: surplus vertices) corresponds exactly to the default number (12) used for conversion of arcs.

As a "sanity check", I consecutively (not cumulatively)highlighted vertices of the mesh in Edit mode. A "sane" vertex highlights orange, along **both** (only two!) edges leading up to (and away from) the selected vertex. Those are fine. In the observed case of surplus vertices and edges, I found vertices, which connect to three (3) edges! Those are apparently "insane".
[image]

[image]

The way I resolved this is, to delete the mesh, then in CAD Sketcher deleted all entities which were connected to the "insane" vertices. Be sure, to also delete any remaining SlvsPoint2D which were part of the deleted entities. Be very sure not to leave any "duplicate" vertices seemingly occupying the same 2D location! If you want to draw a polygon of straight lines only, I found the following approach to work well:  Create all points first, then "snap" to the points when creating lines.

Hope this helps.

Attached are the initial, buggy file plus the second, sane file.

[Zylinderhalter.blend](https://vielzutun.ch/wordpress/public/Zylinderhalter.blend)
[Zylinderhalter2.blend](https://vielzutun.ch/wordpress/public/Zylinderhalter2.blend)