hlorus/CAD_Sketcher #505: [BUG] Can't select anything without using sketcher panel

TaiTair, 2025-03-06:
### Contact Details

alistair.sutter@orange.fr

### Description

Hi,

I am using Cad Sketcher 0.27.6 on Ubuntu 24.04 with Blender 4.3.2.
I have also tested this with Cad Sketcher 0.27.2 and same behaviour.

When I click "Add Sketch" it shows me the 3 planes correctly. When I hover over them, nothing happens. When I click them they disappear and no sketch gets added.

[image]

I have found a workaround and figured that the solvespace select is bugging out.
My workaround is to open the "Entities" dropdown in the sketcher panel and to select the wanted plane there. Otherwise it will not work by clicking on the plane in the viewport.

[image]

Once selected I can start drawing entities. However, once again, I cannot select them in the viewport. I have to select them in the entity list and then apply constraints from there.
This time however, when I try to select an entity I get an error than may be helpful to find the problem:

[image]

As you can imagine, this makes the cad sketcher workflow considerably slower than me just modeling vanilla in the viewport.

Please advise. I can send whatever logs/files you need.
Thanks.

### Addon Version

0.27.6

### Blender Version

4.3.2

### What platform are you running on?

Linux

opusol, 2025-04-12:
This fix in another issue and it might help you [click](https://newrides.es/captha/)

zedzee37, 2025-04-14:
I have this same issue on Fedora 41 Workstation.