hlorus/CAD_Sketcher #514: [BUG] Unable to Install on Blender 4.4.3 using Python 3.13.3

johnnyarojas, 2025-05-11:
### Contact Details

johnny.rojas.me@gmail.com

### Description

CAD Sketcher is not installable on Blender 4.4.3 using Python 3.13.3, when trying to downgrade to the recommended python version by the add-on (Python 3.11), the rest of blender breaks due to incompatibility.

Installation error message:
Extension bl_ext.user_default.CAD_Sketcher is incompatible (This Python version (3.13) isn't compatible with (3.11))

### Addon Version

27.6

### Blender Version

4.4.3

### What platform are you running on?

Linux

rcarmo, 2025-05-16:
Go inside the CAD_Sketcher folder and do:

```bash
pip download py_slvs --dest ./wheels --only-binary=:all: --python-version=3.13 
```

...then update the `blender_manifest.toml` to use that particular wheel it's just downloaded.

zetxx, 2025-07-09:
> Go inside the CAD_Sketcher folder and do:
> 
> pip download py_slvs --dest ./wheels --only-binary=:all: --python-version=3.13 
> ...then update the `blender_manifest.toml` to use that particular wheel it's just downloaded.

did it, now i have module isnt registered:

[image]

zetxx, 2025-07-10:
Hey, newest version of the plugin or module ?

hlorus, 2025-07-10:
Newest version of CAD Sketcher.