hlorus/CAD_Sketcher #517: [BUG] Can't select point with Vulkan backend

ra100, 2025-05-30:
### Description

I'm sorry, not sure if this goes to Vulkan backend report or here.
When I tried Vulkan, looks like lines are narrower and points are actually not visible (or just 1px wide), so unselectable from viewport.

OpenGL
[image]

Vulkan
[image]

### Addon Version

0.27.5

### Blender Version

4.4.3

### What platform are you running on?

Linux

ra100, 2025-06-11:
@sgebbie does this mean, that point handles/points will have to be somehow fixed in Vulkan API?

sgebbie, 2025-06-12:
> However, taking a closer look I am wondering if there might be a problem with the `id_shader_3d` under Vulkan:

On closer inspection, it seems like the shader itself might be fine. But rather the control of the point size is a problem. The result is that the point handles are now `1 x 1` pixels which are difficult to see and pick. This is also affecting the sketch lines, but they are then easier to see.

ra100, 2025-06-12:
Interesting, so maybe change in shaders used?

> The script need to use the point shaders to draw points.